DavidBertet/PowerJeep #1: 56v Ego batteries?

flyingtech, 2024-04-07:
What would I need to adapt a 56v battery from my Ego mower?

DavidBertet, 2024-04-08:
56v would fry basically everything I used here.

You need to put a step down converter in between, so you end up with the 12-18v that can be used for this project

flyingtech, 2024-04-09:
Thank you.  
This is what i'm ordering: 
[](https://www.aliexpress.us/item/3256805491763442.html?spm=a2g0o.cart.0.0.3ea838daQJJufR&mp=1&gatewayAdapt=glo2usa
)
I hope that this will work.

DavidBertet, 2024-04-10:
10A with Max 15A might be a tight light while on load. You might want to replace the fuse by a 15A and put it between the battery and converter, or take a bigger buck converter. 

I'll check what mine uses with an amperemeter with the kids, but just blocking the wheels I see it going up to 22A. My 30A fuse never tripped.