DavidBertet/PowerJeep #2: 2 Motor vehicle

joroosen, 2024-04-09:
I have a 2 motor vehicle (Tesla kids power car).

Can I just connect both motors to the same motor driver or would I need 2 motor drivers?

Thanks for this nice project!!

DavidBertet, 2024-04-10:
Yes, one driver is fine. Mine has actually 2 motors in parallel.

They are connected under the car with a fuse each (original harness), and only 2 wires (red/black) are reaching the electronic

Thanks!

joroosen, 2024-04-10:
Alright! Thanks for the reply....I'll do some ordering and keep this past updated on my progress later :-)

derekclawson, 2024-04-19:
I successfully powered 2 motors with separate drivers wired in parallel. One driver would overload and turn off when climbing a small incline. Puting each motor on its own driver fixed the issue and is working great currently. I am running a 20v 8ah hercules battery from harbor freight.