DavidBertet/PowerJeep #3: unable to load webpage after connecting to wifi

flyingtech, 2024-04-14:
Forgive me if this is something abvious I dont have much experience and this is the first time i have ever used VS code. But im having fun learning.

I have managed to get the code compiled and uploaded to the ESP32.  when i connect to the wifi the phone tells me to sign-in to the network. 
[image]

(this is after entering Rubicon! and connecting.) I get the following page:
[image]

If i click the hambergur at the top right i can sellect "use this network as is
[image]

but when i try to go to the the IP address get this.
[image]

🤷‍♂️ Thank you for the help

P.S. I dont know who is more excited about this project, me or my kids. LOL

DavidBertet, 2024-04-16:
Congrats, you are doing great!

Oh no, I feel bad. I pushed something wrong and fixed it few minutes later. Seems like you got that commit.

Pull the latest code from `main` and you should be good to go!

If not, open PlatformIO "Monitor" in VSCode to see what's going on on the ESP32.

flyingtech, 2024-04-30:
Ok i have verified that i have the latest code according a friend of mine.  see screenshot for the last change made in power_wheel.c line 417.
[image]

but i'm still getting the same issue with too many redirects.  bellow is the log from the monitor.

I (664710) wifi:new:<10,0>, old:<10,0>, ap:<10,2>, sta:<255,255>, prof:10
I (664710) wifi:station: d4:3a:2c:ab:d8:39 join, AID=1, bgn, 20
␛[0;32mI (664770) wifi: station d4:3a:2c:ab:d8:39 join, AID=1␛[0m
W (664930) wifi:<ba-add>idx:2 (ifx:1, d4:3a:2c:ab:d8:39), tid:0, ssn:0, winSize:64
␛[0;32mI (665160) esp_netif_lwip: DHCP server assigned IP to a station, IP is: 192.168.4.2␛[0m
W (665220) wifi:<ba-add>idx:3 (ifx:1, d4:3a:2c:ab:d8:39), tid:7, ssn:2, winSize:64
␛[0;32mI (665260) dns_captive_portal: Received 32 bytes from 192.168.4.2 | DNS reply with len: 48␛[0m
␛[0;32mI (665260) dns_captive_portal: Waiting for data␛[0m
␛[0;32mI (665260) dns_captive_portal: Received 47 bytes from 192.168.4.2 | DNS reply with len: 63␛[0m
␛[0;32mI (665270) dns_captive_portal: Waiting for data␛[0m
␛[0;31mE (665300) webfile: Request received for /generate_204␛[0m
␛[0;31mE (665350) webfile: Failed to stat file: /spiffs/generate_204␛[0m
␛[0;32mI (665360) websocket: WS Client Disconnected 59␛[0m
␛[0;31mE (665650) webfile: Request received for /generate_204␛[0m
␛[0;31mE (665790) webfile: Failed to stat file: /spiffs/generate_204␛[0m
␛[0;31mE (665820) webfile: Request received for /␛[0m
␛[0;31mE (665880) webfile: Failed to stat file: /spiffs/index.html␛[0m
␛[0;31mE (665940) webfile: Request received for /␛[0m
␛[0;31mE (666000) webfile: Failed to stat file: /spiffs/index.html␛[0m
␛[0;31mE (666060) webfile: Request received for /␛[0m
␛[0;31mE (666130) webfile: Failed to stat file: /spiffs/index.html␛[0m
␛[0;31mE (666200) webfile: Request received for /␛[0m
␛[0;31mE (666350) webfile: Failed to stat file: /spiffs/index.html␛[0m
␛[0;31mE (666420) webfile: Request received for /␛[0m
␛[0;31mE (666570) webfile: Failed to stat file: /spiffs/index.html␛[0m
␛[0;31mE (666660) webfile: Request received for /␛[0m
␛[0;31mE (666710) webfile: Failed to stat file: /spiffs/index.html␛[0m
␛[0;31mE (666780) webfile: Request received for /␛[0m
␛[0;31mE (666880) webfile: Failed to stat file: /spiffs/index.html␛[0m
␛[0;31mE (666980) webfile: Request received for /␛[0m
␛[0;31mE (667100) webfile: Failed to stat file: /spiffs/index.html␛[0m
␛[0;31mE (667170) webfile: Request received for /␛[0m
␛[0;31mE (667280) webfile: Failed to stat file: /spiffs/index.html␛[0m
␛[0;31mE (667350) webfile: Request received for /␛[0m
␛[0;31mE (667420) webfile: Failed to stat file: /spiffs/index.html␛[0m
␛[0;31mE (667490) webfile: Request received for /␛[0m
␛[0;31mE (667570) webfile: Failed to stat file: /spiffs/index.html␛[0m
␛[0;31mE (667670) webfile: Request received for /␛[0m
␛[0;31mE (667800) webfile: Failed to stat file: /spiffs/index.html␛[0m
␛[0;31mE (667880) webfile: Request received for /␛[0m
␛[0;31mE (667950) webfile: Failed to stat file: /spiffs/index.html␛[0m
␛[0;31mE (668020) webfile: Request received for /␛[0m
␛[0;31mE (668090) webfile: Failed to stat file: /spiffs/index.html␛[0m
␛[0;31mE (668150) webfile: Request received for /␛[0m
␛[0;31mE (668260) webfile: Failed to stat file: /spiffs/index.html␛[0m
␛[0;31mE (668330) webfile: Request received for /␛[0m
␛[0;31mE (668460) webfile: Failed to stat file: /spiffs/index.html␛[0m
␛[0;31mE (668560) webfile: Request received for /␛[0m
␛[0;31mE (668620) webfile: Failed to stat file: /spiffs/index.html␛[0m
␛[0;31mE (668680) webfile: Request received for /␛[0m
␛[0;31mE (668770) webfile: Failed to stat file: /spiffs/index.html␛[0m
␛[0;31mE (668860) webfile: Request received for /␛[0m
␛[0;31mE (668950) webfile: Failed to stat file: /spiffs/index.html␛[0m
␛[0;31mE (669140) webfile: Request received for /generate_204␛[0m
␛[0;31mE (669340) webfile: Failed to stat file: /spiffs/generate_204␛[0m
␛[0;32mI (669350) websocket: WS Client Disconnected 60␛[0m

flyingtech, 2024-05-05:
UPDATE:
I think the prolem was that i dont have Platfor IO setup correctly for the esp32 board type.  reset platform IO to defualt settings and reflashed the esp32.  and it is now working as expected.

Thank you for all the help!!!  doing new things is always hard but so worth it.  marking this as closed.

closed 2024-05-05 by flyingtech