DavidBertet/PowerJeep #4: 20V PowerPlus battery

joroosen, 2024-04-15:
Hello,

I'm excited to share that all the necessary parts have been ordered and are on their way. I'm eagerly anticipating starting this project!

I have a kids' Tesla Model S with a broken battery that I plan to convert using this GitHub project.

While I have plenty of 18V Bosch Pro batteries, I prefer not to use my power tool batteries for my kids' toys. Therefore, I'm considering using 20V battery packs from the "Power Plus" brand, which are affordable and readily available in my area.

Could there be any potential risks or problems with using 20V instead of 18V? From what I understand, all the components in this project should be capable of handling 20V, correct? I just need to check the datasheets of the motors installed in the Tesla to confirm.

I see great potential in this project. By connecting multiple batteries to increase the milliampere-hour (mAh), we can add a variety of exciting accessories.

I'll keep you updated on my progress and share pictures along the way.

Thank you!

flyingtech, 2024-04-15:
I got my parts on order also.  You could step down the voltage to 18 before the motor driver module if you are concerned about burning it up.  That is what ill be doing with the 56V batteries I have.

DavidBertet, 2024-04-16:
All parts here should work fine with 20V. But it's likely the same batteries.

18V is nominal, 20V is max. For marketing purposes, some are advertising 20V...

In any case, no need to step down here as the components are ok. If it's really 20V, don't go all the way to 100% but max it out at 90% on the interface. You can update that line in the `index.html` so you can't mess up
```
<div class="slider_container">
  <div class="slider_label">Max forward</div>
  <input
    id="maxForwardInput"
    class="slider"
    type="range"
    min="20"
    max="100" <-------------------------------- put 90 here
    value="0"
    oninput="return onSliderInput(this)"
  />
  <div id="maxForwardInputValue">0</div>
</div>
```

DavidBertet, 2024-04-24:
@derekclawson successfully used a 20V battery. I'm closing that issue 👍

closed 2024-04-24 by DavidBertet

rufusmccoot, 2024-04-24:
I also used a 20v battery (from my Worx brand tools). I didn't have to change anything in the software.