DavidBertet/PowerJeep #8: Can we run this on 5v instead of 3.3v?

derekclawson, 2024-04-18:
In looking at the datasheet for the motor driver, it is specifying 5v be the supply voltage. Have you tested feeding the esp32 5v on the VIN pin and to the IBT_2 Motor driver, then using the 3.3 v pin on the ESP32 to run through the pedal and the switch for the signaling?
Would this make any difference or does the motor driver run fine on 3.3v?

thatdecade, 2024-04-18:
It should work as you want.  The BTS 7960 datasheet, section "Electrical Characteristics - Control and Diagnostics", give an examples of typical VINH and VINL values and are not dependent on the VCC voltage.  Same with the 74HC244D buffer chip.  If you are worried, the 3V3 pin on the ESP32 dev board is a 3.3V input & output. You can input 5V to VIN and output 3.3V to the motor driver.

The parts that worry me are:

- The ESP32 circuit you have is probably a clone, and may or may not have VIN/USB diode. So inputting 5V to VIN with the USB connected could break your computer's USB port.
- Why you want to switch from 3.3V to 5V. Assuming you want 5V so you can re-use for both the ESP32/Driver and other Car Sound Effect Electronics, is probably a bad idea.  Sound effects are a big draw, and voltage sags crash the ESP32.

derekclawson, 2024-04-18:
@thatdecade  I appreciate the explanation. I did try running the ESP32 and IBT_2 motor driver on 5v. It works, but there is no change to the issue I was having of the driver shutting off on load. I am currently wiring up 2 drivers in parallel (one for each motor)  to see if that helps.

DavidBertet, 2024-04-24:
Agreed with @thatdecade. While 5V should work, there is not much benefit.

The IBT_2 is known to overheat. I went with thermal paste, but seems like you found a solution by using 2 of them.

I'll try to update the README accordingly, thanks for sharing!