DavidBertet/PowerJeep #17: Error on firmware build - 'panic_emergency_shutdown_hook' undeclared

tobes49, 2025-08-07:
Hello,

I cant get it to complete the firmware build:
✗ Firmware build failed                                                    	 
✗ Build error:
src/main.c: In function 'app_main':
src/main.c:51:33: error: 'panic_emergency_shutdown_hook' undeclared (first use in this function)
   51 |   esp_register_shutdown_handler(panic_emergency_shutdown_hook);
  	|                             	^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
src/main.c:51:33: note: each undeclared identifier is reported only once for each function it appears in
*** [.pio/build/esp32doit-devkit-v1/src/main.c.o] Error 1
========================= [FAILED] Took 216.36 seconds =========================
Environment      	Status	Duration
-------------------  --------  ------------
esp32doit-devkit-v1  FAILED	00:03:36.361
==================== 1 failed, 0 succeeded in 00:03:36.361 ====================

If you want any other info let me know,

Cheers

DavidBertet, 2025-08-09:
Hey @tobes49, thanks for raising it!

I messed up my last commit and missed a local modification. This is fix

tobes49, 2025-08-09:
Just tried and all uploaded correctly this time.
gfamily@gfamily-SATELLITE-C75-A:~/PowerJeep-2-beta/PowerJeep-2-beta$ ./install.sh 

┌─────────────────────────┐
│     PowerJeep CLI       │
│   Setup & Deploy Tool   │
└─────────────────────────┘

ℹ Detected OS: Linux

🚀 Checking for Node.js...
────────────────────────────────────────────────────────────
✓ Node.js found (v18.19.1)
✓ npm found (9.2.0)
🚀 Checking other prerequisites...
────────────────────────────────────────────────────────────
✓ PlatformIO found in: /home/gfamily/.platformio/penv/bin/pio                   
✓ PlatformIO info (PlatformIO Core, version 6.1.18)                             
✓ Project structure verified
✓ Prerequisites check completed!

🚀 Configuring backend...
────────────────────────────────────────────────────────────
ℹ Current board: esp32doit-devkit-v1
? Do you want to change the board? (y/N) n

🚀 Building backend...
────────────────────────────────────────────────────────────
✓ Firmware built successfully                                                   

🚀 Building frontend...
────────────────────────────────────────────────────────────
✓ Frontend dependencies installed                                               
✓ Frontend built successfully                                                   
✓ Files copied to backend/data                                                  

🚀 Uploading via Serial...
────────────────────────────────────────────────────────────
ℹ Make sure your ESP32 is connected via USB and drivers are installed.
? Upload firmware now? (Y/n) y
ℹ Using serial port: /dev/ttyUSB0                                               
ℹ Connected                                                                     
✓ Firmware uploaded successfully                                                
ℹ ESP32 should restart automatically with the new firmware
? Upload frontend (web files)? (Y/n) y
ℹ Using serial port: /dev/ttyUSB0                                               
ℹ Connected                                                                     
✓ Frontend uploaded successfully                                                

────────────────────────────────────────────────────────────
✓ 🎉 Setup and deployment completed successfully!
ℹ Your ESP32 power wheel controller is now running.
ℹ Check your wifi for "PowerJeep" if you haven't set it up yet.

I'm out today but will try it over the weekend to see how it works.  

Thank you

DavidBertet, 2025-08-09:
Awesome! Mine is back to the garage, kids finally melted the gearbox ahah

Taking the opportunity to use a pair of BTS7960, it was getting hot and sometime tripped it

closed 2025-08-09 by DavidBertet

tobes49, 2025-08-13:
Mine does the same sometimes and turns off for a moment. 
 The biggest issue I've been having is sometimes wont reverse.  But I think this is to do with running 12v and 40% of 12v is not enough to start the motors when the battery is running low.

My plan is to change over to a 4s battery which will be a higher voltage and let the vehicle be faster and also reverse correctly. But I have a spare 6s 14000ma battery laying around so tempted to try that.

DavidBertet, 2025-08-16:
Oh feel free to update [the value here!](https://github.com/DavidBertet/PowerJeep/blob/v2-beta/frontend/src/components/tab/ProfilesTab.svelte#L627)

I put 40% because with a 18v battery I didn't want the kids to fly in reverse, but with 12v you can safely go higher.

If it turns back on automatically, it's likely the voltage protection that turns off (XH-M609). The battery sag under load and can momentarily go below the cut off value. The car stops, the battery "recovers" and voltage goes above reactivation, which turns it back on.

When BTS7960 trips, you have to manually turn off/on power. Turns out using 2 is not a solution as they aren't balanced.
I'm currently tuning [the frequency](https://github.com/DavidBertet/PowerJeep/blob/v2-beta/backend/src/power_wheel_controller.c#L150), this makes a huge difference. However with 12v your BTS7960 shouldn't get too hot?